robotology-legacy/yarp-wholebodyinterface #15: yarpWholeBodyModel without yarp ports

hu-yue, 2015-01-19:
Hi, I'm trying to use only the yarpWholeBodyModel for some computations, but it tries to open yarp ports, and thus requires yarp server to be running:
```
RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Cannot connect to port /root
yarp: Port /model/torso/rpc:o failed to activate (invalid address)
yarp: Cannot connect to port /root
yarp: Port /model/torso/command:o failed to activate (invalid address)
yarp: Cannot connect to port /root
yarp: Port /model/torso/state:i failed to activate (invalid address)
yarp: Cannot connect to port /root
yarp: Port /model/torso/stateExt:i failed to activate (invalid address)
yarpdev: ***ERROR*** driver <remote_controlboard> was found but could not open
Problems instantiating the device driver torso
```
I tried to not put the [WBI_YARP_JOINTS] option in the config file but then it gives this error:
```
[ERR] wholeBodyInterface error: joint torso_pitch[ERR]  not found in WBI_YARP_JOINTS section of configuration file
```
Is there a way to use the yarpWholeBodyModel without opening yarp ports?

traversaro, 2015-01-19:
In the new_wbi_ID branch or the old branch?

P.s. we just added the frames, as soon they regenerated the URDF you should get your frames :  )

hu-yue, 2015-01-19:
new_wbi_ID

Thanks for the info =)

traversaro, 2015-01-19:
Can you paste the configuration file that your are using? 
See the migration guide for more information https://github.com/robotology-playground/yarp-wholebodyinterface/blob/new_wbi_ID/Migration.md#wbi-configuration-files .

hu-yue, 2015-01-19:
This is the configuration file, it is in the local folder, as well as the urdf:
```
robot icub
# the urdf_file filename is searched using the ResourceFinder::findFile() method
urdf iCubHeidelberg01.urdf

[WBI_YARP_JOINTS]
#Every "actuated" joint in the wbi is mapped to an axes of a yarp controlboard
#torso
torso_yaw   = (torso,0)
torso_roll  = (torso,1)
torso_pitch = (torso,2)
#left_leg
l_hip_pitch = (left_leg,0)
l_hip_roll  = (left_leg,1)
l_hip_yaw   = (left_leg,2)
l_knee          = (left_leg,3)
l_ankle_pitch   = (left_leg,4)
l_ankle_roll    = (left_leg,5)
#right_leg
r_hip_pitch    = (right_leg,0)
r_hip_roll     = (right_leg,1)
r_hip_yaw      = (right_leg,2)
r_knee         = (right_leg,3)
r_ankle_pitch  = (right_leg,4)
r_ankle_roll   = (right_leg,5)

[WBI_ID_LISTS]
#joints
ICUB_TORSO_JOINTS = (torso_pitch,torso_roll,torso_yaw)
ICUB_LEFT_LEG_JOINTS = (l_hip_pitch,l_hip_roll,l_hip_yaw,l_knee,l_ankle_pitch,l_ankle_roll)
ICUB_RIGHT_LEG_JOINTS = (r_hip_pitch,r_hip_roll,r_hip_yaw,r_knee,r_ankle_pitch,r_ankle_roll)
ROBOT_MAIN_JOINTS = (ICUB_TORSO_JOINTS,ICUB_LEFT_LEG_JOINTS,ICUB_RIGHT_LEG_JOINTS)
```

I took out all the sensors related stuff.

traversaro, 2015-01-19:
Can you try now? I made the option of getting the limits from the real robot optionally, an pushed some documentation (It will be finally generated as soon as we merge new_wbi_ID in master).

hu-yue, 2015-01-19:
Not it's ok, thanks!

closed 2015-01-19 by hu-yue