robotology-legacy/yarp-wholebodyinterface #18: Optimize setControlMode implementation

traversaro, 2015-01-21:
Currently the setControlMode implementation is slow (it was part of a trade off for simplicity and having a fast setControlReference).

randaz81, 2015-01-21:
Beware: on the real robot the low level setControlMode() may take several tens of milliseconds to be executed and the method is intentional blocking.

traversaro, 2015-01-21:
Yes, the point is to implement it using the multi axis methods provided by the IControlMode2 interface.

traversaro, 2020-02-01:
As yarp-wholebodyinterface has been deprecated, we will not solve this issue. Please use the  multiple-axis method of the YARP's IControlMode interface (
`yarp::dev::IControlMode::setControlModes(const int n_joint, const int *joints, int *modes)`, see 
https://www.yarp.it/classyarp_1_1dev_1_1IControlMode.html#a24322b7b0c6e707e7ee29acd2e8f4c05) to efficiently change the control mode of the joints, and make sure there is no bottleneck of this in your YARP robot.

closed 2020-02-01 by traversaro