robotology-legacy/yarp-wholebodyinterface #26: Matlab crashes probably because of computeBaseVelocity

DanielePucci, 2015-03-25:
I'm running matlab and as soon as I launch the simulation, it crashed. By looking at the stack trace, I think that the problem is due to the new implementation of the computeBaseVelocity.

@naveenoid @traversaro @francesco-romano 

Stack Trace (from fault):
[  0] 0x00007f809f0f2c6c                    /lib/x86_64-linux-gnu/libc.so.6+01186924
[  1] 0x00007f8014baabfa /home/shared/build-x86_64/codyco-superbuild/install/mex/../lib/libyarpwholebodyinterface.so.0.2.1+00171002 _ZN7yarpWbi18yarpWholeBodyModel15computeJacobianEPdR$
[  2] 0x00007f8014bba586 /home/shared/build-x86_64/codyco-superbuild/install/mex/../lib/libyarpwholebodyinterface.so.0.2.1+00234886 _ZN7yarpWbi22yarpWholeBodyEstimator19computeBaseVelo$
[  3] 0x00007f8014bbadfa /home/shared/build-x86_64/codyco-superbuild/install/mex/../lib/libyarpwholebodyinterface.so.0.2.1+00237050 _ZN7yarpWbi22yarpWholeBodyEstimator3runEv+00001050
[  4] 0x00007f8013f8b952 /home/shared/build-x86_64/yarp/lib/libYARP_OS.so.1+01243474 _ZN25RateThreadCallbackAdapter3runEv+00000178
[  5] 0x00007f8013fc245c /home/shared/build-x86_64/yarp/lib/libYARP_OS.so.1+01467484 _Z18theExecutiveBranchPv+00000092
[  6] 0x00007f8010619636                           /usr/lib/libACE-6.0.3.so+01037878 _ZN21ACE_OS_Thread_Adapter6invokeEv+00000166
[  7] 0x00007f809f363b50              /lib/x86_64-linux-gnu/libpthread.so.0+00027472
[  8] 0x00007f809f0ad7bd                    /lib/x86_64-linux-gnu/libc.so.6+00903101 clone+00000109

traversaro, 2015-03-25:
As workaround for this issue, I pushed a modification [1] such that the new base and velocity computations are enabled just if the `estimateBasePosAndVel` option is passed to the `yarpWholeBodyState`, otherwise it does not call the `computeBaseVel` and `computeBasePos` option . As soon as we solve this issue, we can go to computing the base quantities by default. 

[1] https://github.com/robotology-playground/yarp-wholebodyinterface/commit/027e933d63e7c070321f92131a2c00f36a2105b6

naveenoid, 2015-03-31:
Fixed in fe3053a80652bf4810e4bf34c9c9e8a57403e29d and subsequent WBI commits.

closed 2015-03-31 by naveenoid