robotology-legacy/yarp-wholebodyinterface #29: Base Pos/Vel Estimation Errors

rlober, 2015-03-30:
I am having some trouble with the new base state estimation code. I am trying to get the base pos and vel each time `thread::run()` is executed and print the linear pos/vel components to the terminal. The values being printed are way off so I was hoping someone could look at my code snippets below and tell me if I went wrong somewhere or if it is a bug in the estimator. 

FYI: I am using the iCub fixed feet model in Gazebo with up to date yarp and icub-main (as of this writing).

**thread.h** 
```hpp
yarp::sig::Vector fb_Hroot_Vector;
yarp::sig::Vector fb_Troot_Vector;
```

**thread.cpp**
```cpp
fb_Hroot_Vector = yarp::sig::Vector(12, 0.0);
fb_Troot_Vector = yarp::sig::Vector(6, 0.0);
```

**in `run()`**
```cpp
bool res_fb_Hroot_Vector = wbi->getEstimates(ESTIMATE_BASE_POS, fb_Hroot_Vector.data());
bool res_fb_Troot = wbi->getEstimates(ESTIMATE_BASE_POS, fb_Troot_Vector.data());
   
std::cout<< "\n---\nfb_Hroot:\n" << fb_Hroot_Vector(0) << fb_Hroot_Vector(1) << fb_Hroot_Vector(2) << std::endl;
std::cout<< "fb_Troot:\n" << fb_Troot_Vector(0) << fb_Troot_Vector(1) << fb_Troot_Vector(2) << "\n---\n";
```

**terminal output**
```
---
fb_Hroot:
ISIRWholeBodyController thread started
6.91484e-3106.91484e-3102.31174e-316
fb_Troot:
6.91484e-3106.91484e-3102.31174e-316
---

...

---
fb_Hroot:
6.91484e-3106.91484e-3102.31174e-316
fb_Troot:
6.91484e-3106.91484e-3102.31174e-316
---
^C[try 1 of 3] Trying to shut down
```

Thanks in advance,
Ryan

naveenoid, 2015-03-30:
Just for confirmation, could you run the test in codyco-modules : 
https://github.com/robotology/codyco-modules/blob/master/src/tests/wholeBodyInterfaceIcubTest/main.cpp
This will verify if you are using the right version of all libraries and modules.

traversaro, 2015-03-30:
@naveenoid what is the expected size of ESTIMATE_BASE_POS ?

naveenoid, 2015-03-30:
its a vector of size 12. First 3 denote position, next 9 are for the rotation matrix corresponding to base orientation.

rlober, 2015-03-30:
Sorry for the delay.

I tried building the test but for some reason I am getting a yarp include compile error `fatal error: yarp/os/Network.h`:

```
ryan@ISIR2014057:~/Code/iCub_Software/codyco-superbuild/libraries/yarpWholeBodyInterface/tests/yarpWholeBodyRootWorldTest$ mkdir build
ryan@ISIR2014057:~/Code/iCub_Software/codyco-superbuild/libraries/yarpWholeBodyInterface/tests/yarpWholeBodyRootWorldTest$ cd build/
ryan@ISIR2014057:~/Code/iCub_Software/codyco-superbuild/libraries/yarpWholeBodyInterface/tests/yarpWholeBodyRootWorldTest/build$ cmake ..
-- The C compiler identification is GNU 4.6.3
-- The CXX compiler identification is GNU 4.6.3
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Using iCub from build tree
-- /home/ryan/Code/iCub_Software/yarp/build/generated_include/home/ryan/Code/iCub_Software/yarp/src/libYARP_OS/include/home/ryan/Code/iCub_Software/yarp/src/libYARP_sig/include/home/ryan/Code/iCub_Software/yarp/src/libYARP_math/include/home/ryan/Code/iCub_Software/yarp/src/libYARP_dev/include/home/ryan/Code/iCub_Software/yarp/src/libYARP_name/include/home/ryan/Code/iCub_Software/yarp/src/libYARP_manager/include/home/ryan/Code/iCub_Software/yarp/src/libYARP_logger/include/home/ryan/Code/iCub_Software/yarp/src/yarpserver/include
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ryan/Code/iCub_Software/codyco-superbuild/libraries/yarpWholeBodyInterface/tests/yarpWholeBodyRootWorldTest/build
ryan@ISIR2014057:~/Code/iCub_Software/codyco-superbuild/libraries/yarpWholeBodyInterface/tests/yarpWholeBodyRootWorldTest/build$ make
Scanning dependencies of target yarpWholeBodyRootWorldTest
[100%] Building CXX object CMakeFiles/yarpWholeBodyRootWorldTest.dir/main.cpp.o
/home/ryan/Code/iCub_Software/codyco-superbuild/libraries/yarpWholeBodyInterface/tests/yarpWholeBodyRootWorldTest/main.cpp:11:29: fatal error: yarp/os/Network.h: No such file or directory
compilation terminated.
make[2]: *** [CMakeFiles/yarpWholeBodyRootWorldTest.dir/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/yarpWholeBodyRootWorldTest.dir/all] Error 2
make: *** [all] Error 2
ryan@ISIR2014057:~/Code/iCub_Software/codyco-superbuild/libraries/yarpWholeBodyInterface/tests/yarpWholeBodyRootWorldTest/build$ 

```

I added `message (STATUS ${YARP_INCLUDE_DIRS})` to the CMakeLists and the output (shown above confirms that CMake locates yarp so I am not sure what the issue is. I am a total Cmake nooooob.

naveenoid, 2015-03-30:
recompile superbuild with the latest commit. @traversaro modified the Cmakelists in the codyco-modules test directory in order to compile the wholeBodyInterfaceIcubTest automatically. Then tell me the test results.

naveenoid, 2015-03-30:
@rlober BTW I just realised you tried to build a different test. Recompile codyco superbuild. The test I mentioned is in codyco-modules. Its called wholeBodyInterfaceIcubTest

rlober, 2015-03-31:
Good morning, @naveenoid @traversaro 

Thanks for the commit.

So I recompiled and ran `wholeBodyInterfaceIcubTest` with icub fixed feet in gazebo. The model didn't explode or anything but the base estimation never changes from zero. *Question:* Does **base** mean world frame or root frame? I was under the impression that it was for the `root_link`, but if this is the case then we should be reading non-zero values for the position estimations at least. If however **base** means world frame then all zeros would make sense. Can either of you two clarify exactly what this should be estimating and how we should be using it?

I have copied the entire terminal output to make debugging a bit easier...  

**terminal output**
```
ryan@ISIR2014057:~/Code/iCub_Software/codyco-superbuild/build/install/bin$ wholeBodyInterfaceIcubTest 
Robot name is given 

[ERR] locomotionControl: impossible to open wholeBodyInterface: wbi_conf_file option missing||| configuring
||| no policy found
||| default config file specified as yarpWholeBodyInterface.ini
||| checking [/home/ryan/Code/iCub_Software/codyco-superbuild/build/install/bin/yarpWholeBodyInterface.ini] (pwd)
||| checking [/home/ryan/.config/yarp/robots/icubGazeboSim] (robot YARP_CONFIG_HOME)
||| checking [/home/ryan/.local/share/yarp/robots/icubGazeboSim] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/icubGazeboSim] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/icubGazeboSim] (robot YARP_CONFIG_DIRS)
||| checking [/home/ryan/Code/iCub_Software/yarp/build/share/yarp/robots/icubGazeboSim] (robot YARP_DATA_DIRS)
||| checking [/home/ryan/Code/iCub_Software/icub-main/build/share/iCub/robots/icubGazeboSim] (robot YARP_DATA_DIRS)
||| checking [/home/ryan/Code/iCub_Software/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim] (robot YARP_DATA_DIRS)
||| found /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim
||| checking [/home/ryan/Code/iCub_Software/yarp/build/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/ryan/Code/iCub_Software/icub-main/build/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/ryan/Code/iCub_Software/codyco-superbuild/build/install/share/codyco/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/ryan/Code/iCub_Software/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/yarpWholeBodyInterface.ini] (robot)
||| found /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/yarpWholeBodyInterface.ini
||| finding file [yarpWholeBodyInterface.ini]
icubWholeBodyInterface created, adding joints
Joints added, calling init method
RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /wbiTestactuator/torso/rpc:o active at tcp://134.157.18.48:10080
yarp: Port /wbiTestactuator/torso/command:o active at tcp://134.157.18.48:10081
yarp: Port /wbiTestactuator/torso/state:i active at tcp://134.157.18.48:10082
yarp: Port /wbiTestactuator/torso/stateExt:i active at tcp://134.157.18.48:10083
yarp: Sending output from /wbiTestactuator/torso/rpc:o to /icubGazeboSim/torso/rpc:i using tcp
yarp: Sending output from /wbiTestactuator/torso/command:o to /icubGazeboSim/torso/command:i using udp
yarp: Receiving input from /icubGazeboSim/torso/state:o to /wbiTestactuator/torso/state:i using udp
yarp: Receiving input from /icubGazeboSim/torso/stateExt:o to /wbiTestactuator/torso/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /wbiTestactuator/left_arm/rpc:o active at tcp://134.157.18.48:10084
yarp: Port /wbiTestactuator/left_arm/command:o active at tcp://134.157.18.48:10085
yarp: Port /wbiTestactuator/left_arm/state:i active at tcp://134.157.18.48:10086
yarp: Port /wbiTestactuator/left_arm/stateExt:i active at tcp://134.157.18.48:10087
yarp: Sending output from /wbiTestactuator/left_arm/rpc:o to /icubGazeboSim/left_arm/rpc:i using tcp
yarp: Sending output from /wbiTestactuator/left_arm/command:o to /icubGazeboSim/left_arm/command:i using udp
yarp: Receiving input from /icubGazeboSim/left_arm/state:o to /wbiTestactuator/left_arm/state:i using udp
yarp: Receiving input from /icubGazeboSim/left_arm/stateExt:o to /wbiTestactuator/left_arm/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /wbiTestactuator/right_arm/rpc:o active at tcp://134.157.18.48:10088
yarp: Port /wbiTestactuator/right_arm/command:o active at tcp://134.157.18.48:10089
yarp: Port /wbiTestactuator/right_arm/state:i active at tcp://134.157.18.48:10090
yarp: Port /wbiTestactuator/right_arm/stateExt:i active at tcp://134.157.18.48:10091
yarp: Sending output from /wbiTestactuator/right_arm/rpc:o to /icubGazeboSim/right_arm/rpc:i using tcp
yarp: Sending output from /wbiTestactuator/right_arm/command:o to /icubGazeboSim/right_arm/command:i using udp
yarp: Receiving input from /icubGazeboSim/right_arm/state:o to /wbiTestactuator/right_arm/state:i using udp
yarp: Receiving input from /icubGazeboSim/right_arm/stateExt:o to /wbiTestactuator/right_arm/stateExt:i using udp
[... cut ...]
(Q, dq, d2q):   24.63 	 -0.00 	 -0.00
BasePos: 0.00 0.00 0.00

BaseRot: 0.00 0.00 0.00
 0.00 0.00 0.00
 0.00 0.00 0.00

(Q, dq, d2q):   24.63 	 -0.00 	 -0.00
BasePos: 0.00 0.00 0.00

BaseRot: 0.00 0.00 0.00
 0.00 0.00 0.00
 0.00 0.00 0.00

Test finished. Press return to exit.
Q:    24.6	 15.6	 14.7	-20.5	 44.8	 15.1	 65.0	 15.0	-20.5	 44.1 15.1	 65.1	 15.0	 12.5	 0.0	-0.0	-9.8	-1.2	 0.0	 12.5	-0.0	 0.0	-9.8	-1.2	-0.0
Test finished. Press return to exit.
Test finished. Press return to exit.
```

Thanks again,
Ryan

naveenoid, 2015-03-31:
@rlober  your assumption is right. It does compute the root link position and rotation and the corresponding velocities. And if you notice in your terminal output, it reads the config file to find the world frame position which is currently fixed as "l_sole" 

However I think I have located your problem. In 027e933d63e7c070321f92131a2c00f36a2105b6, @traversaro inserted an option to stop computing world-base-rototranslation because it gave some realtime crashes in the robot. This is another issue that I just debugged and fixed in the latest commit fe3053a80652bf4810e4bf34c9c9e8a57403e29d. Its related to some code vanishing between merging my feature branch and master.

For now, pull and test the latest commit from codyco superbuild.  your wbi config file (/home/ryan/Code/iCub_Software/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/yarpWholeBodyInterface.ini) must be modified to include 
"estimateBasePosAndVel" under the [WBI_STATE_OPTIONS], i.e. immediately after the line saying "WORLD_REFERENCE_FRAME l_sole".

rlober, 2015-03-31:
Great thank you for the info. I made the changes and everything seems to be working in the test example. When I ran our code immediately after it seemed to be working as well. :+1:

Where in the source code can I modify `/home/ryan/Code/iCub_Software/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/yarpWholeBodyInterface.ini` so that I don't have to remember to change it every time I rebuild codyco-superbuild?

traversaro, 2015-03-31:
here: https://github.com/robotology-playground/yarp-wholebodyinterface/tree/master/app/robots/icubGazeboSim

rlober, 2015-03-31:
Great! I updated the ini files for icubGazeboSim as well as Paris01/Paris02.  Would you like me to commit and submit a pull request or would you prefer to leave this up to the user to add? @traversaro @naveenoid 

This issue has been resolved.

closed 2015-03-31 by rlober

traversaro, 2015-03-31:
I guess if everything is working you can commit the configuration files (even directly to the repo!).

rlober, 2015-03-31:
Done.

rlober, 2015-03-31:
Thanks a bunch guys I appreciate the help and great work on the Base Estimation @naveenoid @traversaro !  Looking forward to next week.