robotology-legacy/yarp-wholebodyinterface #34: [conf] Migrate iCub with FT sensor in the ankles to the new models

traversaro, 2015-04-09:
We recently discovered and fixed a bug in the generation of the model with FT sensor in the ankle. Model have already been update for iCubParis*, but as soon as we can test on the real robots we should migrate iCub v1 with ankle ft sensors (iCubGenova03) and iCub v2.5 (iCubGenova01, iCubGenova03).

traversaro, 2015-04-27:
I really don't remember anything related to this... we should test all the new models as soon as `iCubGenova01` is back. 

cc @DanielePucci (as a remainder)

DanielePucci, 2015-04-27:
Yep.

traversaro, 2015-07-22:
Apparently we already migrated iCubGenova01 and iCubGenova03 in https://github.com/robotology-playground/yarp-wholebodyinterface/commit/b944a0244c5c96950028e8e861f8bb64263e7fc1 . We will now add a model for iCubGenova02 cloned from iCubGenova01 but with an additional mass on the chest, estimated to be ~1.5 kg.

traversaro, 2015-07-22:
Robots model still to verify:
 * [ ] iCubDarmstad01
 * [ ] iCubGenova04

traversaro, 2017-01-25:
`iCubDarmstad01` model was update for sure during the CoDyCo Y3 demo last year, and `iCubGenova04` was transformed in iCubNancy01.

closed 2017-01-25 by traversaro