robotology-legacy/yarp-wholebodyinterface #35: [computeJacobian] Seg fault when computing r_hand jacobian

rlober, 2015-04-27:
When I use `wbi::computeJacobian()` for the `r_hand`, which has index 49 according to `getFrameList`, I get a segmentation fault.

I do not get this fault when I calculate the jacobian for `l_hand`, which has index 33 according to `getFrameList`. 

I am using icub fixed in Gazebo.

1.  Are the frames and their indexes correct?
2.  Could this have something to do with the wbi estimate bas pos/vel or l_sole options?

traversaro, 2015-04-27:
1. The frame serialization is not declared explicitly anyway, and is obtained when loading the urdf model. The id seem anyway correct to me.

traversaro, 2015-04-27:
2. Mh... there was a concurrency problem with yarpWholeBodyModel, but it should have been solved in https://github.com/robotology-playground/yarp-wholebodyinterface/commit/bda55f3428ad5250f54e42e670022965edcbfa7d .

Do you have the code that is causing the segfault ?

rlober, 2015-04-27:
```cpp
robot->computeJacobian(owm_pimpl->q.data(), owm_pimpl->Hroot_wbi, index, owm_pimpl->segJacobian_rm[index].data());
```

traversaro, 2015-04-27:
So  `robot->computeJacobian(owm_pimpl->q.data(), owm_pimpl->Hroot_wbi, index, owm_pimpl->segJacobian_rm[index].data());`
with `index = 33` works, but with `index 49` segfaults .. just to be sure, owm_pimpl->segJacobian_rm[49].data() is properly allocated?

rlober, 2015-04-27:
Yup. Just saw that too. Sorry man.

Just for your pleasure... the vector of jacobians was hard coded to be size [44] which came from summer. I have fixed this to use `robot->getFrameList().size()`. 

Now working. Thanks Silvio!

closed 2015-04-27 by rlober