robotology-legacy/yarp-wholebodyinterface #37: how to tell the wholebodyinterface not to find manually deactivated DOF

mingxing-liu, 2015-06-17:
Hi everyone, 
We have to manually deactivated the left_leg in icub_all because of some communication issues with the left leg of our icub. Could someone please tell me how to make the wholebodyinterface not to trying to find the deactivated DOF (such as those of the left leg), so as to avoid the following errors?

[ERROR]Problem connecting to /icub/left_leg/rpc:i, is the remote device available?
[ERROR]Problem connecting to /icub/left_leg/command:i, is the remote device available?
[ERROR]Problem connecting to /icub/left_leg/state:o from /actuator/left_leg/state:i, is the remote device available?
[ERROR]*** Extended port /icub/left_leg/stateExt:o was not found on the controlBoardWrapper I'm connecting to. Falling back to compatibility behaviour
[WARNING]Updating to newer yarp and the usage of controlBoardWrapper2 is suggested***
yarpdev: ***ERROR*** driver <remote_controlboard> was found but could not open
[ERROR]Problems instantiating the device driver left_leg
[ERR] yarpWholeBodyActuators::openDrivers error: unable to open controlboard left_legof robot icub
...
[ERR] Error while initializing yarpWholeBodyActuators interface.
[ERR] Error while initializing yarpWholeBodyModel interface.
[ERR] Error while initializing yarpWholeBodyModel interface.
[ERR] Error while initializing yarpWholeBodyStates interface.
Error while initializing whole body interface. Closing module

francesco-romano, 2015-06-17:
Hi @mingxing-liu 
All the wholebodyInterface "powered" applications read a configuration file which is specify which joints will be used and configured.
Usually this file is called `yarpWholeBodyInterface.ini`. Try to understand which one is the one opened by your application and modify it accordingly (at the end there are some lists: find the one you use and remove the leg from it)

mingxing-liu, 2015-06-17:
Thank you @francesco-romano

closed 2015-06-17 by mingxing-liu