robotology-legacy/yarp-wholebodyinterface #38: problem in position control mode

mingxing-liu, 2015-06-23:
Hi, 
I'm testing a control program that uses the position control mode (setControlMode(CTRL_MODE_POS, 0, -1)), and then send reference joint positions and joint velocities int thread::run() by setting
        robot->setControlParam(CTRL_PARAM_REF_VEL, ref_vel.data(), ALL_JOINTS);
        robot->setControlReference(position_cmd.data());

When I try the program in gazebo it works.
But when running it on real iCub, the robot does not move until I close the program, when the robot seems to execute the last position command (with motions that seems to be smoothed by minjerk between the initial posture and the last reference posture).

Is there something that I should add to the program so that the control command will be executed at each time step?

Thanks in advance.

traversaro, 2015-06-23:
If you want to send streaming position references to the low level position controller, bypassing the minimum jerk trajectory, please use the `CTRL_MODE_DIRECT_POSITION` control mode . 

If you actually want to use the interpolated control, if you want to send the velocity parameter in all the joints you should use  the call:
~~~
robot->setControlParam(CTRL_PARAM_REF_VEL, ref_vel.data());
~~~
without the third argument, where `ref_vel` is a vector of dimension `getActuatorList().size()`.

mingxing-liu, 2015-06-23:
Setting to CTRL_MODE_DIRECT_POSITION makes the control works. Thank you @traversaro.

closed 2015-06-23 by mingxing-liu