robotology-legacy/yarp-wholebodyinterface #39: setControlMode with CTRL_MODE_POS does not work as expected?

hu-yue, 2015-07-10:
Hi,
I have a doubt in the use of ```setControlMode(wbi::CTRL_MODE_POS, &ref, j)```.
I thought that if I make this call with the same reference value, the position as per reference should be kept, but instead it's accumulating and increasing every time. e.g. ref = 30, the first call the joint moves to 30, the second to 60 and so on.
Also, is it necessary to use a ```setControlReference``` after a ```setControlMode``` or not?
Because it seems that in the case of ```CTRL_MODE_TORQUE``` the torque has been set to the reference (0) without using ```setControlReference```, while in the case of ```CTRL_MODE_POS``` the joint did not move at all (ref = 30), while it did move with ```setControlReference```, but does what described above.

This happens when using it on a specific joint, while using it on all the joints ```setControlMode(wbi::CTRL_MODE_POS)``` it behaves as I thought it should (not accumulating reference).

Everything with the model in Gazebo (with gazebo-yarp-plugin master branch, not sea-joints).

traversaro, 2015-07-10:
The basic idea is that you do once, before you start a controller, a:
~~~
setControlMode(wbi::CTRL_MODE_POS, &ref, j).
~~~
The main purpose of this command (that is quite slow) is to ensure that the control mode of the underlying low-level loop is set to POSITION mode. You provide an (optional) `ref` value just to avoid that the loops starts with a unknown reference value. 

Once you do that, in the "control loop" you are supposed to just use the `setControlReference` to set the reference value. 

The accumulating behavior is quite strange... can you try with `CTRL_MODE_DIRECT_POSITION`? Given that you are on the simulator is worth trying.

hu-yue, 2015-07-10:
Ok, thanks.
Now I modified the code so that the control mode is set at the beginning only, then using ```setControlReference```. However on position it's still accumulating.

I tried to stream the same positions with ```CTRL_MODE_DIRECT_POSITION```, the robot did a very fast motion.

hu-yue, 2015-07-10:
Ok solve because there was a misunderstanding of the units.
References should be in radians and not degrees and I did not get this.

closed 2015-07-10 by hu-yue

traversaro, 2015-07-10:
Additional documentation pushed in https://github.com/robotology-playground/wholebodyinterface/commit/812bbf7811de9ff53e5b348691bac20f17fbcffb .