robotology-legacy/yarp-wholebodyinterface #43: Troubles in defining the YARP_ROBOT_NAME with icubGazeboSim

kzts, 2015-10-29:
I found it misleading that all robots are called iCubCity01, iCubCity02, ... (where iCub is properly spelled with capital C) while icubGazeboSim is spelled with small c. I configured the environment variable YARP_ROBOT_NAME with capital C and nothing was working :sob: . I realized I needed to define `export YARP_ROBOT_NAME=icubGazeboSim` and finally everything seems to work :smile: .

traversaro, 2015-10-29:
You are right, that is definitively confusing. 
A  proper naming (also in the sense of `YARP_ROBOT_NAME` ) for the Gazebo models was discussed in https://github.com/robotology-playground/icub-gazebo/issues/16 , but then the proposed changes still need to be implemented.

traversaro, 2020-02-01:
This issue will be tracked in https://github.com/robotology/icub-gazebo/issues/16 .

closed 2020-02-01 by traversaro