robotology-legacy/yarp-wholebodyinterface #49: SensorsMeasurement as the yarpWholeBodySensorsBuffer

naveenoid, 2016-01-04:
Since the iDynTree SensorList and SensorsMeasurement classes (https://github.com/robotology/idyntree/blob/master/src/sensors/include/iDynTree/Sensors/Sensors.hpp) are now functional and can be built from URDF, I think its time we move towards yarpWholeBodySensors simply populate a SensorsMeasurement object (using the setMeasurement method) instead of multiple vectors corresponding to each type of sensor (i.e. the buffers for each kind of measurement). The wholeBodyEstimator class (or any other component) can then simple gain access to a reference to this SensorsMeasurement object. The high level design change would involve iDynTree incorporation in yarpWholeBodyInterface.

This issue will track your comments and the design of the new yarpWholeBodySensors
@traversaro @francesco-romano @jeljaik

jeljaik, 2016-01-06:
@naveenoid You should start by following this: https://github.com/robotology/yarp-wholebodyinterface/issues/42 as we had already started discussing here https://github.com/robotology/codyco-modules/issues/146#issuecomment-159638415, but please please please please please, try to keep all these issues referenced in the dedicated one, i.e. https://github.com/robotology/codyco-modules/issues/146#issuecomment-159638415

naveenoid, 2016-01-06:
I agree, but this is a design level topic. It requires a fundamental change in how wholeBodySensors is working. It requires replacing all the existing vector buffers with a single iDyntree object.

jeljaik, 2016-01-06:
Yes, it's fine, I'm just talking about referencing the issue. By the way, if you need some sample data, for example for the accelerometer I had attached to the foot, you can find it in `$CODYCO_SUPERBUILD_ROOT/main/codyco-modules/src/modules/wholeBodyEstimator/data`

naveenoid, 2016-01-06:
Excellent, thanks! A commit and some sample code coming soon in my PR. I
appreciate your effort to keep the issues tied together.

traversaro, 2018-08-23:
Since this design, we went through several design iteration for our estimators. I think we agree that this issue is  no longer relevant.

closed 2018-08-23 by traversaro