robotology-legacy/yarp-wholebodyinterface #50: Document how the accelerometers are published on Eth iCubs

traversaro, 2016-02-05:
We should properly document how mtb accelerometers are published in yarp ports on ETH iCub. 

For CAN robots, this documentation is provided in : 
http://wiki.icub.org/wiki/Distributed_Inertial_sensing .

There is also a stub of  a section to describe how the accelerometers are published on Ethernet robots, we need to populated that section as discussed in https://github.com/robotology/yarp-wholebodyinterface/issues/42#issuecomment-168395775 .

@nunoguedelha @naveenoid @jeljaik 

If anybody needs credentials to edit the iCub wiki, feel free to contact @mbrunettini .

jeljaik, 2016-02-05:
In the meantime I have collected the most relevant information regarding what I've been doing lately as an entry in [codyco-modules wiki](https://github.com/robotology/codyco-modules/wiki/External-Inertial-Sensors-for-iCubGenova02)

nunoguedelha, 2016-02-05:
Thanks guys!

naveenoid, 2016-02-05:
@jeljaik  :+1:  I will update the wiki with the port names to correspond to the existing table with the CAN robot MTB sensor info.

nunoguedelha, 2016-02-21:
@jeljaik , in the wiki you mentioned [codyco-modules wiki](https://github.com/robotology/codyco-modules/wiki/External-Inertial-Sensors-for-iCubGenova02), you describe the format of the data read from port `icub/right_leg/inertialMTB` as : 
```
a = (n  6.0  (a1  b1 t1 x1 y1 x1)   .... (an  bn tn xn yn xn))
ai = pos of sensor ... see enum type
bi = accel (1) or gyro (2)
```
Could you clarify what `tn` is exactly? any kind of relevant timestamp?
Thanks

naveenoid, 2016-02-22:
i believe its the standard yarp timestamp (uS since last epoch)..especially
when I check the data. Maybe @jeljaik can correct me?

jeljaik, 2016-02-22:
Yes that's correct @naveenoid . I updated the wiki accordingly @nunoguedelha

traversaro, 2018-08-23:
This is not relevant anymore, as the publication of the accelerators and the distributed inertial sensing on iCub  is being migrated to use the Multiple Analog Sensors interfaces in YARP.

closed 2018-08-23 by traversaro