robotology-legacy/yarp-wholebodyinterface #55: Wrong predicted measurements on forearm accelerometers "1b7", "1b8", "1b9", "2b7", "2b8", "2b9"

nunoguedelha, 2016-08-15:
@traversaro, Just a heads-up in case someone uses the `iDynTree::predictSensorsMeasurements` function on the forearm accelerometers **"1b7", "1b8", "1b9", "2b7", "2b8", "2b9"**.
#### The predicted measurements on these sensors will be wrong if we rotate the `*_wrist_prosup` joint or if its angle is not 0 :
The URDF models including the frames for these sensors don't have the Pronosupination joint (`l_wrist_prosup` and `r_wrist_prosup`), and these models define the accelerometers frames as attached to the `l/r_elbow_1` link which is between the elbow and the forearm. While we rotate the `*_wrist_prosup` joint, this link won't rotate with the forearm cover, where the sensors are attached. As `iDynTree::predictSensorsMeasurements` rely on the model to predict the measurements, there is a mismatch between the expected and the real sensors orientation.

nunoguedelha, 2016-08-15:
I guess this should be corrected as soon as the new models including pronosupination are integrated.

traversaro, 2018-08-23:
@nunoguedelha  I guess this was solved a long time ago, right?

closed 2018-08-23 by traversaro