robotology-legacy/yarp-wholebodyinterface #56: Provide configuration files for Walkman robot

traversaro, 2016-09-08:
To make sure that we can use the `iDynTree`, `yarp-wholebodyinterface` and `codyco-modules` software with the Walkman robot, we need to write some configuration files and make sure that the `URDF` files provided upstream work nicely. 

If you have the rights to see, the URDF model for the Walkman can be obtained at: 
* https://gitlab.robotology.eu/walkman-drc/iit-bigman-ros-pkg

Current issues are: 
* [x] Understand the used Walkman's `YARP_ROBOT_NAME` : it is apparently `bigman` .
* [ ] Walkman's. `gazebo-yarp-plugins` configuration files are outdated and do not work with YARP 2.3.66, fixed by @gabrielenava in https://gitlab.robotology.eu/walkman-drc/iit-bigman-ros-pkg/issues/10 . 
* [ ] Make sure that information on F/T sensors are correctly loaded from URDF.

traversaro, 2016-09-08:
For loading the F/T information from URDF, some useful pointers to iDynTree documentation are: 
* [How to load F/T sensors in new iDynTree-pure classes](https://github.com/robotology/idyntree/blob/master/doc/model_loading.md)
* [How to load F/T sensors in old legacy KDL classes](https://github.com/robotology/idyntree/blob/master/doc/model_loading_legacy.md)
Note that currently in `wholeBodyDynamicsTree` the old legacy classes are used, while in the new `wholebodydynamics` YARP device the new classes are used. For this reason all the iCub models in this repo contain the F/T information in both formats.

traversaro, 2016-09-08:
cc @DanielePucci

traversaro, 2016-09-10:
By the way, apparently we discussed how to model F/T sensors back in 2013 with @arocchi in https://github.com/EnricoMingo/iit-coman-ros-pkg/issues/13 . At that time we made the discussion for the Coman, but those decision was inherited by the Walkman model.
The main problem  that prevented us to model the Walkman F/T sensors as fixed joints was that fixed joints where not supported by SDF/Gazebo. 
Fortunately fixed joints simulation has been supported in Gazebo since version 6.0 . 
Unfortunately `robotology-superbuild` is currently supporting Gazebo 5. 
Fortunately `ros-indigo-gazebo6-*` packages seems to be available, so upgrading to a new version of Gazebo seems doable. 
However, it may be worth moving this discussion to a separate issue for clarity.

traversaro, 2016-09-10:
Issue specific to get the external wrenches estimation to work on Walkaman : https://github.com/robotology/codyco-modules/issues/200 .

gabrielenava, 2016-09-12:
Just to keep in mind what has been done until now, I did a .md file that documents all my work with Walkman. It is available [here](https://github.com/gabrielenava/BackupWalkman) and I will keep it more or less updated (I think I will update it once a week).

traversaro, 2016-09-12:
Great!

gabrielenava, 2017-01-25:
Also the last point is done, because the information from F/T sensors have been loaded in Simulink, verified and used for Yoga ++ demo.

traversaro, 2017-01-25:
@gabrielenava ok, then the issue can be closed.

closed 2017-01-25 by traversaro