robotology-legacy/yarp-wholebodyinterface #61: [demo-codyco branch] compilation error link to yarp-wholebodyinterface

misaki43, 2016-11-24:
Hello, 
I need to use the demo-codyco branch to be able to launch the stand up demo with icubGazeboSim, but there are some compilation errors when I build it:
[errorCompilation.txt](https://github.com/robotology/yarp-wholebodyinterface/files/610981/errorCompilation.txt)
Can you help me to solve it please?

@DanielePucci @paraschos @serena-ivaldi

traversaro, 2016-11-24:
I just update the branch in https://github.com/robotology/yarp-wholebodyinterface/commit/7e75410fd4fce6ed6ed21bd467344a1754a2c488 , can you try again? Thanks.

paraschos, 2016-11-24:
@traversaro perfect, it compiles now. 

One question: I had the demo_codyco and I did make update-all in superbuild. As a result it pulled the master in it and not the demo_codyco. It was not complicated to resolve it, but how is it done properly?

misaki43, 2016-11-24:
@traversaro yes it is working, thanks a lot !

@paraschos I let you close the issue when you will have finished

traversaro, 2016-11-24:
@paraschos  sorry, I was not aware of the fact that the chair simulation setup was being already disseminated to partners, when we have time with @gabrielenava we should provide a smoother workflow. 

The options are: 
##  Let the superbuild use the new branch
Modify the [`cmake/ProjectsTags.cmake`](https://github.com/robotology/codyco-superbuild/blob/master/cmake/ProjectsTags.cmake) in the `codyco-superbuild` repo, that is then included in 
https://github.com/robotology/codyco-superbuild/blob/master/CMakeLists.txt#L74 . In this way  you can specify the desired branch/tag/commit for each `<subproject>` by setting the `<subproject>_TAG` CMake variable. 
In your case the `ProjectsTags.cmake` content would be:
~~~
set(codyco-modules_TAG demo_codyco)
~~~
The `ProjectsTags.cmake` file can be just kept modified locally or pushed to a branch of `codyco-superbuild` itself. 
In this way the update step of the superbuild will use the specified branch.

## Instruct the superbuild to ignore the projects on which you are working
If you want that the superbuild `update` step does not update a subproject, you can just set the `YCM_EP_DEVEL_MODE_<SUBPROJECT>` CMake variable of the superbuild to `ON`.  
For more on this, see http://robotology.github.io/ycm/gh-pages/master/manual/ycm-superbuild.7.html#developer-mode .

paraschos, 2016-11-24:
@traversaro thanks Silvio!

closed 2016-11-24 by paraschos