robotology-legacy/yarp-wholebodyinterface #70: yarp-wholebodyinterface does not compile (for an iDynTree problem) on Ubuntu 14.04 with CMake 3.5

traversaro, 2017-03-24:
@liesrock @diegoferigo @gabrielenava @DanielePucci

traversaro, 2017-03-24:
@liesrock Actually I was wrong. 
iDynTree exports the C++11 flags **just** on GCC < 5 (so in Ubuntu 14.04, see logic in https://github.com/robotology/idyntree/blob/master/src/core/CMakeLists.txt#L127 ) because as iDynTree now uses C++11 internally, not exporting the C++11 flags in GCC < 5 would result in a runtime ABI mismatch. In GCC >= 5 and Clang there is no problem in compiling iDynTree using C++11 and the downstream libraries using C++98 because the ABI are compatible. Then the `yarp-wholebodyinterface` library is compiled with C++11 also on Travis, and so this is not the difference that is causing our problems.

traversaro, 2017-03-24:
The problem was much simpler. 

Even if `USE_SYSTEM_YARP` (see https://robotology.github.io/ycm/gh-pages/master/manual/ycm-superbuild.7.html#ycm-superbuild-manual-for-developers) was set to `FALSE`, the `yarp-wholebodyinterface` was finding the `YARP` from the [`robotology-superbuild`](https://github.com/robotology-playground/robotology-superbuild) that was installed in the system. 
This caused the `include` directory of the install folder of the `robotology-superbuild` to be passed to the compiler during the compilation of `yarp-wholebodyinterface`. Unfortunately, while  `yarp-wholebodyinterface` was correctly finding the `iDynTree` present in the codyco-superbuild, another (older) version of iDynTree was present in the `robotology-superbuild`, and so the iDynTree's headers from the old iDynTree version in the `robotology-superbuild` were included in place of the new ones, present in the install of the `codyco-superbuild`. 
Manually pointing `yarp-wholebodyinterface` to the YARP in the `codyco-superbuild` solved the problem, but I don't know if it is something that will happen again in the future on a similar setup.

closed 2017-03-24 by traversaro

DanielePucci, 2017-03-24:
Nice catch @traversaro !

liesrock, 2017-03-24:
Thanks @traversaro ! I cleanup the system env variables by not sourcing the robotology-setup.bash
Now we can compile the codyco-superbuild on a clean system.

DanielePucci, 2017-03-24:
👍