robotology-legacy/yarp-wholebodyinterface #72: Not able to compile this repo with codyco-superbuild

gabrielenava, 2017-04-21:
I downloaded `codyco-superbuild` repo and I tried to compile it, but I got the following error (that seems related to yarpWholeBodyInterface):

```
[ 10%] Building CXX object CMakeFiles/yarpwholebodyinterface.dir/src/yarpWholeBodyInterface.cpp.o
In file included from /home/gnava/Software/codyco-superbuild/libraries/yarpWholeBodyInterface/include/yarpWholeBodyInterface/yarpWholeBodyInterface.h:23:0,
                 from /home/gnava/Software/codyco-superbuild/libraries/yarpWholeBodyInterface/src/yarpWholeBodyInterface.cpp:19:
/home/gnava/Software/codyco-superbuild/libraries/yarpWholeBodyInterface/include/yarpWholeBodyInterface/yarpWholeBodyActuators.h:144:21: error: ‘IOpenLoopControl’ is not a member of ‘yarp::dev’
         std::vector<yarp::dev::IOpenLoopControl*>     iopl;
                     ^
/home/gnava/Software/codyco-superbuild/libraries/yarpWholeBodyInterface/include/yarpWholeBodyInterface/yarpWholeBodyActuators.h:144:21: error: ‘IOpenLoopControl’ is not a member of ‘yarp::dev’
/home/gnava/Software/codyco-superbuild/libraries/yarpWholeBodyInterface/include/yarpWholeBodyInterface/yarpWholeBodyActuators.h:144:49: error: template argument 1 is invalid
         std::vector<yarp::dev::IOpenLoopControl*>     iopl;
                                                 ^
/home/gnava/Software/codyco-superbuild/libraries/yarpWholeBodyInterface/include/yarpWholeBodyInterface/yarpWholeBodyActuators.h:144:49: error: template argument 2 is invalid
CMakeFiles/yarpwholebodyinterface.dir/build.make:86: recipe for target 'CMakeFiles/yarpwholebodyinterface.dir/src/yarpWholeBodyInterface.cpp.o' failed
make[5]: *** [CMakeFiles/yarpwholebodyinterface.dir/src/yarpWholeBodyInterface.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/yarpwholebodyinterface.dir/all' failed
make[4]: *** [CMakeFiles/yarpwholebodyinterface.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/yarpWholeBodyInterface.dir/build.make:108: recipe for target 'libraries/yarpWholeBodyInterface/CMakeFiles/YCMStamp/yarpWholeBodyInterface-build' failed
make[2]: *** [libraries/yarpWholeBodyInterface/CMakeFiles/YCMStamp/yarpWholeBodyInterface-build] Error 2
CMakeFiles/Makefile2:1573: recipe for target 'CMakeFiles/yarpWholeBodyInterface.dir/all' failed
make[1]: *** [CMakeFiles/yarpWholeBodyInterface.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```
FYI: on my pc, both `yarp` and `icub-main` are in the `devel` branch.

francesco-romano, 2017-04-21:
I am afraid that this [PR](https://github.com/robotology/yarp/pull/1145) completely removed the open loop control mode.

traversaro, 2017-04-21:
Crap, I forgot about this.

traversaro, 2017-04-21:
I am on it.

closed 2017-04-21 by traversaro