robotology-legacy/yarp-wholebodyinterface #82: Deprecate yarp-wholebodyinterface

traversaro, 2017-08-29:
We are currently removing `yarp-wholebodyinterface` from being a dependency on WB-Toolbox (see https://github.com/robotology/WB-Toolbox/issues/46). 
Given the current state of the library and the fact that most of its functionality have been migrated to `iDynTree::KinDynComputations` iDynTree class and to the `remotecontrolboardremapper` YARP device, it would be great to deprecate completely the `wholebodyinterface` and the `yarp-wholebodyinterface` to avoid to have to maintain them. 
The tasks necessary to solve before we are able to deprecate the library are:  
* Migrate WB-Toolbox to use  `iDynTree::KinDynComputations` iDynTree class and the `remotecontrolboardremapper` YARP device (in progress in https://github.com/robotology/WB-Toolbox/issues/46 )
* Migrate the vvv repositories: 
   * https://github.com/vvv-school/tutorial_impedance-control
   * https://github.com/vvv-school/assignment_computed-torque
  to use  `iDynTree::KinDynComputations` iDynTree class and  the `remotecontrolboardremapper` YARP device (to be prepared for vvv18) 
* Discuss with ISIR the status of https://github.com/ocra-recipes/ocra-wbi-plugins and if it make sense see if we can find human resource to migrate it to use  `iDynTree::KinDynComputations` iDynTree class and  the `remotecontrolboardremapper` YARP device
* Move all the URDF models installed by the yarpWholeBodyInterface . Some of them will be part of the new https://github.com/robotology-playground/icub-models , but for some of them (`icubGazeboSim`, `iCubHeidelberg01`) we need to understand what to do. 
* Migrate the `mex-wholebodymodel` repository (possibly changing its name): https://github.com/robotology/mex-wholebodymodel . I think a shim based on the matlab bindings of iDynTree::KinDynComputations should be a relatively easy job.

DanielePucci, 2017-08-29:
@traversaro thanks a lot for this. Shouldn't we migrate also the modules in https://github.com/robotology/codyco-modules/tree/master/src/modules that use the wbi and we want to keep maintaining ?

traversaro, 2017-08-30:
Yep. I forgot about those. At this moment, **none** of those modules are actively maintained. They compile, but there is no automatic test checking that they are working correctly, and they are not used regularly on the simulator or the robot, so if they stopped working we would have no way to know, so I considered all of them to be deprecated. If there is any interest in maintaining any of this, we should plan for their migration.

traversaro, 2018-02-02:
Related: https://github.com/vvv18-dynamics/vvv18-dynamics.github.io/issues/1 .

traversaro, 2020-04-05:
All the mentioned points have been addressed.

closed 2020-04-05 by traversaro