robotology-legacy/yarp-wholebodyinterface #84: Move icubGazeboSim urdf model

diegoferigo, 2017-12-01:
The deprecation of the yarp-wholebodyinterface is almost finalized, and there are still some decision to take.

Currently for using the icub-gazebo model for simulation, the urdf taken by WB-Toolbox 2.0 is stored in this repository which will slowly disappear. We should find a new location for it and all the related ini files.

@gabrielenava @traversaro @nunoguedelha @S-Dafarra

traversaro, 2018-08-23:
I though about it, and I think the proper solution would be to generate a URDF from the models contained in `icub-gazebo` (without hands because it is the model that is has associated with `icubGazeboSim` historically) and commit and install that one in the `icub-gazebo` repo. In the meanwhile, we can also just move the urdf model contained in here in the `icub-gazebo` repo.

diegoferigo, 2019-04-25:
The new model used in Gazebo is `iCubGazeboV2_5`, hosted in https://github.com/robotology-playground/icub-models. The `icub-gazebo` is still working, even though is not anymore maintained.

Closing.

traversaro, 2019-04-25:
The `icubGazeboSim` model is currently used in some vvv-school tutorials, and in general it is (unfortunatly) quite used. I still think that 
>  In the meanwhile, we can also just move the urdf model contained in here in the icub-gazebo repo.
is probably the best option.

On the other hand, I think we can drop `iCubHeidelberg01` unless @hu-yue is aware of anyone working on it.

diegoferigo, 2019-04-25:
> The icubGazeboSim model is currently used in some vvv-school tutorials

Good catch. Let's keep this pending for some more while then

traversaro, 2020-02-01:
PR for fixing this issue open at https://github.com/robotology/icub-gazebo/pull/73 .

closed 2020-02-01 by traversaro